maxritter/diy-thermocam #103: TP4057 Charging module no longer available

rossoe, 2023-01-16:
Hi
I'm searching for TP4057 and I've only seen stock of below item in India with no international shipping available.
[image]

This is the only similar part I can find in UK on ebay:
[image]
But it's holes are in different locations so would not match the pins on motherboard.

What is best option? Can above be hacked to fit with wires rather than relying on pins?

Gilabite, 2023-01-16:
I used a TP4056, no mods to the board needed. these are what I used https://www.amazon.com/gp/product/B00LTQU2RK/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

rossoe, 2023-01-16:
Ah yes I just spotted someone mentioning the use of TP4056 here - https://github.com/maxritter/DIY-Thermocam/issues/66
They gave a little detail re the schematic:
[image]

I'm fairly new to electronics so just to confirm is this how it should end up - I assume the cables to the rear LIPO holes ensures it makes use of reverse polarity protection on charging board?:
[image]

maxritter, 2023-05-06:
There is indeed no way to source the TP4057 anymore. Thanks for the instructions how to make it work with the TP4056. I have included this as the default part in the partlist now.

closed 2023-05-06 by maxritter

rossoe, 2023-11-23:
ok a little update as I just finished putting it together and now have what I think is the correct setup, this keeps all the wiring on one side of the pcb, as there was not much room between case and pcb to have wires around the back connecting the lipo battery as per discussions in this [thread](https://github.com/maxritter/diy-thermocam/issues/66).
The wires to the LED also allowed me to get both green and blue working which reflects both 'charging' and 'full' states for battery.
[image]

[image]

maxritter, 2023-11-25:
@rossoe Thanks for the detailed update with the images and I am glad it works this way!

chrisp429, 2023-12-06:
Do you think TC4056 with usb-C would suit? I would prefer to use a usb-C cable instead of mirco USB.
Eg: 
[image]
like this one:
https://www.amazon.it/Aideepen-Interfaccia-ingresso-caricabatterie-protezione/dp/B0BZS73V5T/
or

rossoe, 2023-12-06:
I've used that one, however as you can see from the pics above the intended charging slot facing out is the Micro USB on the teensy 4.1 microcontroller which has corresponding hole cut out on the case, so unless you use the whole unit without a case it's not really practical to rotate the charge controller board and make a new hole in the case for it.

chrisp429, 2023-12-06:
> I've used that one, however as you can see from the pics above the intended charging slot facing out is the Micro USB on the teensy 4.1 microcontroller which has corresponding hole cut out on the [...]

Thank you rossoe, it is correct, I hanen't assembled one yet, so I was confused :).
Chris