maxritter/diy-thermocam #104: Lepton 3.1R Compatibility

griffin175, 2023-09-26:
Hey Max,

It looks like the new Lepton 3.1R is not yet compatible with the diy-thermocam firmware.

When using a Lepton 3.1R, the boot-up fails. The self diagnostic menu shows "FLIR Lepton failed" and continuing to boot results in the error message "Lepton error, try to reset".
My unit works fine with the Lepton 3.5.

The new part number is 500-0758-03. 
The only functional difference from the 3.5 is the wider viewing angle at 95°

griffin175, 2023-10-19:
Max helped me find a temporary work around. You can trick the firmware into thinking the 3.1R is a Lepton 3.5. If you follow this fix and want to switch back to a Lepton 3.5 you will need to undo the change and re-flash the firmware.

Using Platform IO with VS Core:
[https://platformio.org/install/ide?install=vscode](Installing Platform IO)
 
Clone the firmware repo and open the folder in VS Code.
Navigate to 3.0/src/hardware/lepton.cpp and find the `bool lepton_version()` function.
[image]

Change `"05-070170"` to `"05-070850-"` and save.

Connect to the DIY-Thermocam's Teensy via USB and upload the code (top right drop down menu).
After compiling and uploading, the DIY-Thermocam should reboot and start working.

maxritter, 2023-10-20:
@griffin175 I have included the 3.1R HW revision on the latest firmware 3.0.4: https://github.com/maxritter/diy-thermocam/releases/tag/3.0.4

Please flash and check if this works for you

griffin175, 2023-10-21:
Issue solved, thanks Max!

closed 2023-10-21 by griffin175