maxritter/diy-thermocam #115: Using cam without display?

dreiduke, 2024-12-23:
Waiting since a monat for the LCD,
Is it possible to use the cam without display?
When its connected to the computer, thermocam live viewer recognizes the cam.
But says, put it in the live mode.

I tried to push the button on the cam, short and long, no change.
Cannot press any button within live viewer.

maxritter, 2024-12-26:
Hello @dreiduke,

yes this is possible. There is the checkNoDisplay function inside the diagnostics that runs after the device boots. If it detects that there is no LCD controller and no Touch controller attached, it will automatically switch into USB Serial Mode: https://github.com/maxritter/diy-thermocam/blob/679eee969cb9601244d535ed68b883560b834834/firmware/3.0/src/hardware/connection.cpp#L53

If you have VS Code installed on your PC, you can add some print statements to the function to see if it is executed: https://github.com/maxritter/diy-thermocam/tree/master/firmware

When the Device is in USB Serial Mode, it should be detected by the Thermal Live Viewer (https://github.com/maxritter/diy-thermocam/blob/master/software/thermal_live_viewer/README.MD)

Let me know if this helps!

dreiduke, 2024-12-26:
Danke dir,

Und schöne restliche Weihnachten, und nen guten rutsch.

Werd das breakoutboard erstmal an nem Raspi probieren.
Denn der thermal live viewer, lässt mich nix drücken, solang die cam nicht in den serial mide geht, (ich mein den com10 belegt die Cam) Eventuelle print commands werd ich erst später probieren können, habe Dienst, über die Feiertage.

In diesem Sinne, und danke für die rückmeldung, die schnelle

Lg dreiduke alias Falko Wagner
Austria

androjet, 2025-07-12:
Hi!

Can the self-diagnostic results be viewed without a display? 

My device seems to be detected by the Thermal Live Viewer but I have no image. 

Thanks for your time!

maxritter, 2025-07-13:
@androjet: Yes, it's also possible to fetch this information via Serial Connection: https://www.diy-thermocam.net/docs/serial/ (0x7F - getDiagnostic)

maxritter, 2025-07-14:
@androjet Nothing special about the serial communication, it inits at 115200 baudrate (https://github.com/maxritter/diy-thermocam/blob/771a51fd7ccbe0f914daafdd735f9575367b5861/firmware/3.0/src/hardware/hardware.cpp#L535) and then you will be able to send the commands from the serial protocol and/or receive Serial.println() commands.

When I was on Windows I always used the application hTerm to debug Serial stuff, this one worked for sure ;)

androjet, 2025-07-20:
> Waiting since a monat for the LCD, Is it possible to use the cam without display? When its connected to the computer, thermocam live viewer recognizes the cam. But says, put it in the live mode.
>
> I [...]

I had the exact issue. I was getting stuck at "put camera in live mode!" because the flir wasnt connecting properly.

In addition to commenting out some of the display while() loops, it was really helpful to add a while() loop with a serial print command that shows the lepton error code. You can find all that in the check hardware functions. My lepton board was very finicky and this allowed me to fix the connection with live feedback.

Once I got the error code to return as "0" consistently, the live viewer properly ran the diagnostic and displayed the flir image.