maxritter/diy-thermocam #116: Help Needed, When I power on the display is all white and nothing ever comes on display

Atlantis3, 2025-01-29:
Hii everyone
I build the thermocam with all the instructions carefully and when i loaded the firmware the display powers on but then nothing happens, it just stays white all the time. There is no logo or anything.

I then connected the device to Thermocam Live Viewer, it says the device connected and then asks me to put the device in live mode.

Also just something i noticed is that the Lepton closes and opens the shutter when I Power it ON

Could someone help where the problem might be or is my display defective

Thank you very much in advance

maxritter, 2025-01-29:
Hey @Atlantis3,

welcome to the project, let's get your device to work! Please use the conncection checker of your multimeter and verify the data pins of the display connector are all properly routed to the respective pins on the Teensy 4.1: https://www.diy-thermocam.net/docs/hardware/

Also verify the 3.3V and 5V voltage for the display are properly working. The Lepton shutter trigger on start is normal and expected, so at least the Lepton has proper power supply.

Best,
Max

Atlantis3, 2025-01-29:
Hey hii @maxritter 
Thank you very much for your answer. The problem was with my soldering on teensy. I replaced it with other one that had good soldering and everything now worked out properly and the device boots up.

Thanks
Akram

maxritter, 2025-01-29:
Glad you found the solution that quickly! Enjoy your DIY-Thermocam!

closed 2025-01-29 by maxritter