maxritter/diy-thermocam #119: Help needed troubleshooting Lepton error, try to reset . .

Graffics, 2025-03-01:
Hi all,

Need some advice regarding how to continue troubleshooting a newly built thermocam v3.

Had a look through all existing issues and only found one other mention of someone having a similar problem here: https://github.com/maxritter/diy-thermocam/issues/98#issuecomment-1646591422

Purchased a Lepton 3.5 (SKU: 500-0771-01) and a breakout V2 from Digikey along with the linked Display from buydisplay.
Sourced equivalent but not identical switches / buttons locally
Other parts I had on-hand (TP4056) or purchased locally (i.e. Teensy 4.1)

Assembled more or less according to the build instruction and it does indeed work.
However it does not work flawlessly.

Seemingly somewhat random (however can be somewhat influenced) the display shows 'Lepton error, try to reset . .'
At lower battery charge it's almost permanently on that error screen immediately from boot.
However even at max charge it can occur either within a few seconds or after a minute or two, or sometimes not at all. 
It can, rarely, recover from this error itself, but most of the time requires to be power-cycled to recover.

With the battery charge symptoms, I suspected the boost converter was not able to keep up with demand. 
However I've checked both the 5V and 3.3V rails, both are very stable and don't change when the error occurs.

Replacing the battery with a lab bench supply does not resolve the issue, however it does change the behaviour when simulating low battery voltages.
If tested at the lower end of the battery charge voltages, it behaves completely randomly like at max battery charge.
It also behaves with the same seemingly random nature throughout the entire battery voltage range.
Sometimes working for 20+ minutes, most of the time usually errors out within 1 minute.

The only additional thing to note, whether on the bench supply or the fully charged lipo pack, the error usually occurs immediately on first power on in some time and can be 'resolved' temporarily by turning it off and back on in rapid succession.

The consumption of the thermocam only jumps up to 800-900 mA at ~4v (battery) when the Lepton cycles the shutter. Which the genuine Pololu U3V16F5 should be able to handle with ease.
Battery info: 3.7V LiPo 2000mAh rated at 2C continuous discharge with cabling suited to 1A continuous.

Is the best next course of action to desolder the boost converter and power it all from an external 5v source?
Appreciate any assistance in how to continue on from here.

Thanks

Atlantis3, 2025-03-05:
Hey hii
I have a similar issue, for me the lepton does not even start once and shows me the same error every time i start it :(
Were you able to solve the issue ?
Thanks

Graffics, 2025-03-06:
Since posting my original issue, I'd say my situation has degraded into a similar scenario. It's very rare to see anything besides the lepton error message.

I have continued troubleshooting, without any concrete success so far. 

Though I am able to confirm that the Lepton breakout and module are not faulty at least.
Spent most of yesterday painstaking putting together and testing different software solutions on a spare Raspberry Pi connected to the Lepton breakout.
Was able to get valid output from pyleton from the specific lepton3-dev branch: https://github.com/groupgets/pylepton/tree/lepton3-dev
And further to that, was able to get a continuous stream out of https://github.com/groupgets/LeptonModule/tree/master/software/raspberrypi_video once the Qt4 dependency was resolved - Manually compiled Qt4.8.7 as per this answer: https://raspberrypi.stackexchange.com/questions/147356/how-to-compile-qt-4-8-7-under-debian-bookworm-and-bullseye
Took several hours to compile overnight on the Pi

With it not being the the Lepton hardware, I've starting digging into the source code. Nothing so far
My C++ is fairly rusty, however i've found the error message and have been adding additional message printouts to see where it's getting caught. 

Given both pyleton and the raspberrypi_video software do not utilize the i2c interfaces, it might be something in that initialization? However not sure yet.

Will keep working on it, if anyone else can jump in and lend a hand regarding troubleshooting the code / creating debug breakpoints etc. It would be greatly appreciated!

maxritter, 2025-03-06:
Hey @Graffics,

sorry to hear you run into those troubles and thanks a lot for providing the detailed problem description, this already helps a lot. I know how much of a pain it is to get those things compiled on a Raspberry Pi, but now you at least know that neither the Lepton Breakout Board nor the Lepton Module is faulty.

From what I read so far, my first guess what could be the reason for the unstable connection between Lepton and Teensy is a faulty PCB with the SPI lines being affected. Those signals run at a very high frequency and a slight issue on one of those lines can cause timing problems, resulting in the behaviour that you have described.

What you could test it to create direct wire connections between the Lepton and the Teensy by soldering short wires between the affected pins for the SPI protocol: https://github.com/maxritter/diy-thermocam-website/blob/master/static/images/device/schematic.jpg

Maybe this is not enough and you also need to cut the connection to the PCB for those lines somehow, but I would first try things that are not destructive ;)

Let me know if that works!

Best,
Max

Graffics, 2025-03-11:
Hey Max, 

Thanks for the assistance! 
After reading through your suggestion I thought we might be on to something - specifically with how the symptoms were presenting during my initial testing. Thought potentially having the battery in-between the PCB and display may have been the difference between it working sometimes and not at all.
Upon re-testing yesterday, with and without the battery 'installed' - it unfortunately didn't make any difference.

I've followed your suggestions regarding directly wiring between the Teensy and Lepton for the SPI pins - Specifically the Lepton breakout pins: Pin7 (SPI_CLK), Pin9 (SPI_MOSI), Pin10 (SPI_CS) and Pin12 (SPI_MISO)

After soldering direct wires on the back of the PCB (display side) between these pins - First test did give some hope, got about 10 seconds of video output before erroring out on the first boot with the direct wiring without breaking the traces. However after that first boot it would still go immediately to the error state.

After that, I broke all 4 traces where they were accessible and tried again. Without any further luck unfortunately.
I was only able to break the traces in roughly the middle of their runs, so it's possible that they are still interfering? Happy to grab a photo or two to show my handiwork if needed. 
I do have spare PCBs (given MOQ of 5 boards) that I could properly isolate the traces from both sides if you think that'd help? 
Would have to de-solder and re-do the entire board though. 

Thanks for the help so far!
Cheers

maxritter, 2025-03-13:
Hey @Graffics,

sorry to hear your attempts have not been sucessful so far! If you have some spare PCBs left, I would suggest to desolder whatever is possible and start with a fresh one with isolated traces for the Lepton SPI pins, using direct wires instead.

I really hope this will allow you to finally use the DIY-Thermocam in full working mode!

Best,
Max

Graffics, 2025-03-25:
Hey @maxritter,

Appreciate the assistance so far! Apologies for the delay in getting back to you.
I've finally completed an entirely new board as suggested - without any change in results unfortunately.

I broke the same 4 traces as above (SPI_CLK, SPI_MOSI, SPI_CS and SPI_MISO) within a few millimeters of the pads on both sides. So as to prevent any antenna-like behaviour for EMI etc. and then re-connected those pads via direct wiring.
Confirmed the traces were broken on either side with a multi-meter prior to re-connecting them.
Then De-soldered and re-populated the new board and completely re-assembled without much issue.
Confirmed continuity of all pins

I did also take the opportunity to test an external 5v supply instead of connecting the Polulu boost module.
This did not seem to make a difference either, tested both with and without the boost module after re-assembly. 

The symptoms are unfortunately the same, the device boots directly to the lepton error message.
To note, I have observed that the boot time for the device is noticeably slower when it boots directly into the error message than if it was to boot in a working state (Though I haven't seem it boot into a working state for a while)

Any further ideas / troubleshooting steps?
Again, if photos would help I'd be happy to provide - just let me know what you'd be after
Appreciate your help again thus far

Cheers

seantronsen, 2026-01-18:
@Graffics were you ever able to get this figured out? Over the couple of weeks in my spare time, I've been assembling, disassembling, and debugging my build but haven't had much luck in getting past the lepton error screen. 

I've confirmed the sensor unit is fine, essentially replicating your raspi testing steps. Using my multimeter, I've also confirmed each of the connections from the teensy microcontroller to the breakout board, with all checks passing. Some additional notes on my end: 
- using lepton 3.1R
- tried two different lepton breakout boards: purethermal from groupgets and breakoutboard v2. main differences are 10 vs 20 pins respectively and differing pin placements. 
- confirmed pinouts, traces, etc across schematics and using the current autosoft eagle free edition.
- sd card fails out some times.
- lepton always fails, but the boot diagnostic is not always the same. if I remove the sd card, it will mark lepton as OK. if I re-insert, sometimes the lepton is marked as fail. other times the diagnostic is skipped (assuming that means all OK)

next planned steps: 
- verifying communications with oscilloscope
- software debugging
- another full re-assembly on a fresh pcb (though I'd like to avoid this if possible)

I'm doing this in my spare hours on Sundays, so progress will probably be measured on the order of weeks.